raspberrypi/pico-micropython-examples #7: There is a problum with GPIO19 for micropython

CaptainJackey, 2021-01-29:
It can't control high or low with machine.Pin with GIOP19, no higt or low output, but 18, 20 is OK!  please cheak!

Test code as follow:
`
from machine import Pin
import time

LED1 = Pin(19, Pin.OUT)

while True:

    LED1.high()
    time.sleep_ms(1000)
    LED1.low()
    time.sleep_ms(1000)
`

lurch, 2021-01-29:
Duplicate of https://github.com/raspberrypi/pico-sdk/issues/28 which is fixed by https://github.com/raspberrypi/pico-sdk/pull/31 , so this should hopefully be fixed in the next release of MicroPython.

CaptainJackey, 2021-01-29:
> Duplicate of [raspberrypi/pico-sdk#28](https://github.com/raspberrypi/pico-sdk/issues/28) which is fixed by [raspberrypi/pico-sdk#31](https://github.com/raspberrypi/pico-sdk/pull/31) , so this should hopefully be fixed in the next release of MicroPython.

So when will the official website UF2 File will be updated? https://www.raspberrypi.org/documentation/pico/getting-started/

aallan, 2021-01-29:
As soon as possible after the fix has propagated through the repos and all the PRs resolve themselves. We're expecting to do a new SDK release shortly which should resolve the issue. I can then build a new MicroPytyhon binary.

aallan, 2021-01-29:
Looks like the PR just got merged into MicroPython mainline,https://github.com/micropython/micropython/pull/6791#issuecomment-769832231. Not sure that includes the fix though.

dpgeorge, 2021-01-31:
> Not sure that includes the fix though.

It doesn't include it.  We are using the master branch of pico-sdk which doesn't yet have the GPIO19 fix.

lurch, 2021-02-01:
@dpgeorge https://github.com/raspberrypi/pico-sdk/releases/tag/1.0.1 has just been pushed :slightly_smiling_face:

dpgeorge, 2021-02-02:
This should now be fixed in MicroPython mainline, see https://github.com/micropython/micropython/pull/6826

aallan, 2021-02-02:
Nightly builds available from https://micropython.org/download/rp2-pico/

closed 2021-02-02 by aallan