raspberrypi/pico-micropython-examples #9: Example with neopixel_ring does not work correcty with RGBW

mjlongfl, 2021-01-31:
In order to get the example working correctly with RGBW rings, line 12 should be adjusted to use a pull_thresh=32 (instead of 24).

JoeHut, 2021-02-01:
Hey @mjlongfl, did you get it running with only this change? I have a neopixel ring with 12 RGBW LEDs and it doesn't seem to work at all...

mjlongfl, 2021-02-02:
@JoeHut Yes; I have a 16x RGBW ring, which matches the source. I wired it exactly as described in the [wiring diagram](https://github.com/raspberrypi/pico-micropython-examples/blob/master/pio/neopixel_ring/neopixel_ring.png).

As I am a software guy getting into hardware, I don't know if I have any particularly valuable insight. Maybe verify that lines 8-9 (NUM_LEDS and PIN_NUM) match your setup.

johannstark, 2023-08-08:
Hey Guys! i manage to make it work as @mjlongfl said. Should we create a small PR to include a comment in the code or something?

aallan, 2023-08-08:
We accept PRs!