raspberrypi/pico-micropython-examples #10: cant take input

Tank1334, 2021-02-01:
on the example `blink` if i try taking input it freaks out. i have to completly restart it each time.

aallan, 2021-02-01:
Sorry, I don't really understand what you're trying to do here, or the problem you are seeing? The blink program 

```
from machine import Pin, Timer

led = Pin(25, Pin.OUT)
tim = Timer()
def tick(timer):
    global led
    led.toggle()

tim.init(freq=2.5, mode=Timer.PERIODIC, callback=tick)
```

blinks the on-board LED on and off, there is nothing to input?

Tank1334, 2021-02-02:
if i was to do something like:
`from machine import Pin, Timer

led = Pin(25, Pin.OUT)
tim = Timer()
def tick(timer):
    global led
    led.toggle()
    input("TEST")

tim.init(freq=2.5, mode=Timer.PERIODIC, callback=tick)
`
it will return error

aallan, 2021-02-02:
Hey, this sounds like a general MicroPython support question rather than a question specifically about the example code in this repo. I'm going to direct you over to the https://github.com/micropython/micropython/issues repo as the RP2040 port has now been merged into the upstream repo.

closed 2021-02-02 by aallan