raspberrypi/pico-micropython-examples #13: machine.ADC(4) - initialization becoming corrupted

arajnak, 2021-02-02:
sensor_temp = machine.ADC(4) - initialization becoming corrupted by subsequent port allocations e.g.
Vsys = machine.ADC(0)

Will result in temp reading of 84 C instead of 24 C from ADC(4)
When done in the reversed order placing ADC(4) last, it works fine.

# sensor_temp = machine.ADC(4)    # initialization corrupted by subsequent adc channel declarations
Vsys = machine.ADC(0)
Vbus = machine.ADC(1)
Vout = machine.ADC(2)
sensor_temp = machine.ADC(4)      # works

aallan, 2021-02-02:
Is this a general bug report for MicroPython in general, or a bug in a specific piece of example code in this repo?

arajnak, 2021-02-02:
This is a general bug report for the MicroPython machine module implementation.

aallan, 2021-02-02:
We can reproduce the bug. Can you file an issue upstream? /cc @dpgeorge

arajnak, 2021-02-02:
> Reply to: raspberrypi/pico-micropython-examples <reply@reply.github.com>

Thanks, will do.