raspberrypi/pico-micropython-examples #18: pio_blink.py and pio_irq.py examples uses 1000Hz frequency which is too low for Pi Pico

kevinjwalters, 2021-02-20:
The examples https://github.com/raspberrypi/pico-micropython-examples/blob/master/pio/pio_irq.py and https://github.com/raspberrypi/pico-micropython-examples/blob/master/pio/pio_blink.py set a frequency of `1000Hz`.

If my understanding is correct then the maxium divisor is 65535 + 255/256 which at typical 125MHz gives 1907.349Hz. That makes these examples _misleading_ as that frequency cannot be achieved and is almost wrong by a factor of 2.

Is 2000Hz as used by https://github.com/raspberrypi/pico-micropython-examples/blob/master/pio/pio_1hz.py a better choice for low frequency for examples? That's precisely attainable at 125MHz and close at 133Mhz (1.47% over).

richardm292, 2021-03-13:
Just checking as I my have raised a duplicate issue...

I though the clock gets divided by 65535 first and then was further adjusted by the fractional part allowing ( still very approximate) lower clock counts. 

2000 is still the lowest accurate frequency. 

More reading for me to do.

lurch, 2021-03-13:
@kevinjwalters Yes, I've just been looking into this and if you request a PIO frequency of 1000Hz I believe you'd actually get a frequency of 2102.1 Hz  (see my comment on https://github.com/micropython/micropython/issues/7025 for a demonstration of how the maths works). So changing them to _actually_ run at 2000Hz will probably have very little noticeable impact on their output :slightly_smiling_face:

kevinjwalters, 2021-03-13:
My point wasn't about exact accuracy it was about providing _example code_ which uses a value _below_ the lower limit which is misleading as developers will see that value and think it's achievable.

lurch, 2021-03-13:
Yup, which PR #26 fixes...

PR #26 (2021-04-08): Change a couple of PIO examples to run at 2000Hz
Fixes #18
commit 29214c78 (2021-04-08): Change a couple of PIO examples to run at 2000Hz
Fixes #18
closed 2021-04-08 by aallan