raspberrypi/pico-micropython-examples #19: UART on Pico with MicroPython has no documentation, please at least include example

7west, 2021-02-20:
I have scoured for some reference or documentation for the use of UART with MicroPython on the Pico with no luck. The "Raspberry Pi Pico Python SDK" loosely references the MicroPython documentation when discussing UART (pg 14). However, MicroPython's documentation on UART does not seem to apply to the Pico (https://docs.micropython.org/en/latest/library/machine.UART.html), because I don't even think the init() function got ported to the Pico.

I am really lost. I even looked through the "Get started with MicroPython on Raspberry Pi Pico" book and found no examples or references.

Maybe I am being really dumb and missing the documentation on this port. Thank you in advance.

lurch, 2021-02-22:
Yeah, unfortunately the [MicroPython docs](http://docs.micropython.org/en/latest/index.html) haven't been fully updated to reflect the RP2040 port yet https://github.com/micropython/micropython/issues/6855

I've not tried using the Pico's MicroPython UART myself, but I recall seeing [this issue](https://github.com/micropython/micropython/issues/6851) which might give you some useful pointers / example code?

ukscone, 2021-02-22:
i've been playing around with the uart on the raspberry pico's micropython port hooking up an esp8266

the following code works, it's not great (came from the previous thread)

```
from machine import UART, Pin
from time import sleep_us

class myUART(UART):
    def readUntil(self, termination, maxlen=-1, includeTermination=True):
        result = ''
        while maxlen < 0 or len(result) < maxlen:
            if self.any():
                #print("here")
                result += chr(self.read(1)[0])
                #print(result)
                if result.endswith(termination):
                    if not includeTermination:
                        result = result[:-len(termination)]
                    break
            sleep_us(10)
        return result

uart = myUART(0, baudrate=9600, tx=Pin(0), rx=Pin(1), bits=8, parity=None, stop=1)

uart.write("AT+GMR\r\n")
print(uart.readUntil('OK',maxlen=-1, includeTermination=True))
```

giving
```

>>> %Run -c $EDITOR_CONTENT
AT version:1.7.4.0(May 11 2020 19:13:04)
SDK version:3.0.4(9532ceb)
compile time:May 27 2020 10:12:22
Bin version(Wroom 02):1.7.4
OK
>>> 
```

one gotcha was that although it worked at 115200 I was losing characters now and again if more than 20 or so characters being returned. i could probably have dropped the baudrate only a little but i dropped it all the way to 9600 and no problems

aallan, 2021-02-23:
Hey, since we don't have *any* example UART code, I've added this as an example to the repo and will pull it into the Python SDK book, see 873ab463942bc44c57c0fa097ae07605b7971416.

7west, 2021-02-23:
I think it is a decent example, but it is very different from the others. It only has two UART methods, where an example like i2c.py or spi.py has several. The example above also defines a class. Many people new to the Pico and/or Python probably do not know that "self.any()" refers to UART. No offense @ukscone, I just think the official example for UART should be more "Python basic" and cover more UART functionality.

aallan, 2021-02-23:
>  I just think the official example for UART should be more "Python basic" and cover more UART functionality.

We take pull requests.

7west, 2021-02-23:
Understood. Give me a day please.

ukscone, 2021-02-23:
>  No offense @ukscone, I just think the official example for UART should be more "Python basic" and cover more UART functionality.

none taken as I agree with you. i was just giving an idea of what actually currently works rather than what is (not) currently documented as what is currently in the micropython docs doesn't actually work :)

in the mentioned thread earlier in this thread there are a couple of example-y type things that use "plain" micropython as it currently works on the pico that might be better to flesh out

ukscone, 2021-02-23:
not sure if this is any better as an example but it has no class now (in both meanings of the word :) ) but it works and also added the ability to have multiply terminators (i'm a python n00b so please be kind about that bit :) ) that seems to do what I expect

```
from machine import Pin, UART
from time import sleep_us
uart = UART(0,baudrate=9600, tx=Pin(0), rx=Pin(1), bits=8, parity=None, stop=1)

def readUntil(uartObject, termination, maxlen=-1, includeTermination=True):
    terminatedFlag=False
    result = ''
    while maxlen < 0 or len(result) < maxlen:
        if uartObject.any():
            result += chr(uartObject.read(1)[0])
            for terminal in termination:
                if result.endswith(terminal):
                    terminatedFlag=True
                    if not includeTermination:
                         result = result[:-len(terminal)]
                    break
            if(terminatedFlag==True):
                return result
        sleep_us(10)
    return result

print(uart.write("AT+GMR\r\n"))
print(readUntil(uart,['OK','ERROR']))
```

7west, 2021-02-23:
In the same format and spirit as the I2C and SPI examples, I think this is a decent example that covers the basic functions of UART on the Pico:

```python
from machine import UART, Pin

uart1 = UART(1, baudrate=9600, tx=Pin(8), rx=Pin(9), bits=8, parity=None, stop=1)
uart1.write(b'UART on GPIO8 & GPIO9 at 9600 baud\n\r')

uart0 = UART(0)
uart0.write(b'UART on GPIO0 & GPIO1 at 115200 baud\n\r')

rxData = bytes()
while uart0.any() > 0:
    rxData += uart0.read(1)

print(rxData)
```

closed 2021-02-25 by aallan