raspberrypi/pico-micropython-examples #20: pio/pio_spi.py: methods of PIOSPI are missing `self`

kopchik, 2021-02-21:
Hi!

It seems methods here are missing `self`:
https://github.com/raspberrypi/pico-micropython-examples/blob/master/pio/pio_spi.py

kopchik, 2021-02-21:
I think the `pio_spi.py` has a bug: the PIO code expects SS pin which it toggles (line 11, the `.side(0x2)` part), but this pin is not passed here: https://github.com/raspberrypi/pico-micropython-examples/blob/master/pio/pio_spi.py#L30 . So, I think, the MOSI pin acts as SS pin.

lurch, 2021-02-22:
> It seems methods here are missing `self`:

Yup, seems like you're right. AFAICT the method-declarations should be:
 * ` def write_blocking(self, wdata):`
 * `def read_blocking(self, n):`
 * `def write_read_blocking(self, wdata):`

> I think the pio_spi.py has a bug: the PIO code expects SS pin which it toggle

I'm not very familiar with the low-level details of SPI or PIO, so I suspect this is one for @Wren6991 - but I _think_ that as `sideset_base=Pin(pin_sck)` and there are 2 items in the `sideset_init=`, then setting `.side(0x2)` would actually set `Pin(pin_sck+1)` high.

I also see that https://github.com/raspberrypi/pico-micropython-examples/blob/master/pio/pio_spi.py#L30 has `sideset_base=Pin(pin_sck), out_base=Pin(pin_mosi), in_base=Pin(pin_sck)` whereas I _suspect_ that `in_base` ought to be `in_base=Pin(pin_miso)`? It's possible that this is early-version micropython-example code which slipped under the radar in our frenzy to get everything released :laughing: 

There's a more complete example of SPI-implemented-with-PIO at https://github.com/raspberrypi/pico-examples/blob/master/pio/spi/spi.pio

kopchik, 2021-02-24:
> There's a more complete example of SPI-implemented-with-PIO at https://github.com/raspberrypi/pico-examples/blob/master/pio/spi/spi.pio

Great link, this is where I took "inspiration" for my own version of SPI that just does sending, without MISO or CS pins (because I didn't need them for my application). This saved me two pins :).

leycec, 2021-12-08:
**Necro-bump.** We just hit this ourselves! In theory, the `self` portion of this issue should be a trivial fix. Likewise, `in_base=Pin(pin_miso)` is *almost* certainly correct but requires actual hands-on QA. Kinda seems like someone just theory-crafted this entire example in their heads without testing anything. :disappointed:

aallan, 2021-12-08:
The RP2040 MicroPython port has been merged upstream to the main MicroPython repo, see micropython/micropython#6791. Going forward you should report problems with the port there, https://github.com/micropython/micropython/. Technical support for MicroPython problems can be found on the MicroPython forums, https://forum.micropython.org/viewforum.php.

closed 2021-12-08 by aallan