raspberrypi/pico-micropython-examples #27: Need FIFO joining example

IndustrialMuffin, 2021-03-31:
While going through the RP2040 Datasheet [https://datasheets.raspberrypi.org/rp2040/rp2040-datasheet.pdf](url) Chapter 3 PIO, section 3.5 Functional Details, then looking for documentation and examples of FIFO joining in micropython, the only thing I've found is [https://forum.micropython.org/viewtopic.php?t=9814&p=54996](url)

It'd be convenient to have an example of FIFO joining here with all these other wonderful examples.

lurch, 2021-04-01:
That forum thread you linked to says "MicroPython doesn't provide an API for this (yet)."

AFAIK the examples in this repo only illustrate using MicroPython APIs; so without MicroPython API support for joining PIO FIFOs, there won't be any pico-micropython-examples showing how to join PIO FIFOs.

IndustrialMuffin, 2021-04-21:
Seems like a fairly valid point lurch.  Now it looks like fifo_join support has recently been added to MicroPython. Gotta figure out how to use it and test it out. [fifo_join support added here](https://github.com/micropython/micropython/commit/f842a40df4d26bd74b92a3903096fc6b1709222d)

lurch, 2021-04-21:
If you do figure it out, Pull Requests are always welcome :slightly_smiling_face:

kevindawson, 2021-05-02:
@lurch @IndustrialMuffin 

from MicroPython GitHub

issue:
- rp2 PIO: Allow merging Tx/Rx FIFOs together [#6854](https://github.com/micropython/micropython/issues/6854)

pull:
- rp2/rp2_pio: Add fifo_join support for PIO. [#7093](https://github.com/micropython/micropython/pull/7093)

lurch, 2021-05-02:
@kevindawson Yes, that's related to @IndustrialMuffin 's "it looks like fifo_join support has recently been added to MicroPython" comment.

My "Pull Requests are always welcome" comment was WRT to adding an example of using FIFO joining to this `pico-micropython-examples` repo. Unfortunately all the RPi engineers are too busy with other stuff to have time to add new example code.

kevindawson, 2021-05-03:
@lurch 
from MicroPython GitHub

example: [pio_uart_rx.py](https://github.com/micropython/micropython/blob/master/examples/rp2/pio_uart_rx.py)

lurch, 2021-05-03:
I'm not sure if there's value in duplicating example-code between multiple repos? :shrug: (it can often lead to the situation where fixes made in one version don't appear in the other version)

Sounds like a decision for @aallan to make :smiley:

kevindawson, 2021-05-03:
@lurch with all my reading around - example I had seen ;)
Please do not duplicate 
I think it would be better just to have links

lurch, 2021-05-03:
> I think it would be better just to have links

Good idea, see #34 :slightly_smiling_face:

kevindawson, 2021-05-03:
@lurch Cool ;)
@IndustrialMuffin if your a happy bunny now, please close.

closed 2021-05-04 by IndustrialMuffin