raspberrypi/pico-micropython-examples #31: Wrong pins on sh1106 spi example

Garfius, 2021-04-25:
Been trying for 2h+, and finally discovered the SPI wiring pins on the comments are wrong.

Instructions goes as (wrong):
` Sample code sections
 ------------ SPI ------------------
 Pin Map SPI
   - 3v - xxxxxx   - Vcc
   - G  - xxxxxx   - Gnd
   - D7 - GPIO 13  - Din / MOSI fixed *1
   - D5 - GPIO 14  - Clk / Sck fixed *2
   - D8 - GPIO 4   - CS (optional, if the only connected device)
   - D2 - GPIO 5   - D/C
   - D1 - GPIO 2   - Res
`

But only works if you use (right):
`
 Pin Map SPI
  -  GPIO 11  - Din / MOSI fixed *1
  -  GPIO 10  - Clk / Sck fixed *2
`

Also may include a suggestion to use a logic level adapter, because there are 5v sh1106 displays.

[Link to photo of working set, also using lg. lvl. adap. 5v](https://drive.google.com/file/d/1eg0GuBSoJ74SwnFV_cGSm3jrLUeV73sB/view?usp=sharing)

The mistake is located at:
[https://github.com/raspberrypi/pico-micropython-examples/blob/master/i2c/1106oled/sh1106.py](https://github.com/raspberrypi/pico-micropython-examples/blob/master/i2c/1106oled/sh1106.py)

Thanks 4 the great work.

lurch, 2021-04-26:
ping @robert-hh (whose name appears in `sh1106.py`)

Took me a little while to figure out what was going on here, as https://github.com/raspberrypi/pico-micropython-examples/tree/master/i2c/1106oled only shows an I2C wiring-example. But then I figured out that this issue is referring to the code **comments** in sh1106.py rather than any actual example code? So then I managed to figure out that the SPI pins used are actually "allocated" by the line: `# spi = SPI(1, baudrate=1000000)`. So then I found up https://github.com/micropython/micropython/blob/master/ports/rp2/machine_spi.c#L44 which confirms that for SPI interface 1, the default MOSI pin used by MicroPython is GPIO11 and the default SCK pin is GPIO10.

Which matches up with what @Garfius has written above :grin: 

I'll give it a couple of days to see if Robert responds, and if he doesn't I'll fix up the SPI-related pin-number code-comments in https://github.com/raspberrypi/pico-micropython-examples/blob/master/i2c/1106oled/sh1106.py myself :slightly_smiling_face:

robert-hh, 2021-04-26:
Thanks for the Ping. Actually, the driver is a pretty old one, initially made for a ESP8266 set-up. So the pin numbers in the example at the header of the driver code are for this device. In a repository for Rp2040 it should indeed match the pins of a rp2 board. Since I do not own that repo, I can only make a PR for a change, which I will do.

robert-hh, 2021-04-26:
About the suggestion of note on 5V devices. That's not needed since:
- the logical section of the sh1106 chip itself runs at 3.3V, and so it matches the RP2040.
- and even if it were a 5V device: Only  inputs of that module are driven by the RP2040. So there is no risk of a damage. Eventually the the high level of the RP2040 could be to low to be recognized by a 5V device. But I never had problems.

robert-hh, 2021-04-26:
@lurch I made that PR. The example code in the sh1106.py header is tested and works for both SPI and I2C.

closed 2021-04-26 by aallan