raspberrypi/pico-micropython-examples #35: "__thonny_helper isn't defined"  error

biodives, 2021-05-16:
I made a silly coding mistake but the error message threw me off so I hope documenting it here will help others or allow a fix to get a more informative error message. Here is the error message:

```
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 8, in Bomb
NameError: name 'self' isn't defined
>>> THONNY FAILED TO EXECUTE COMMAND get_globals

SCRIPT:
__thonny_helper.print_mgmt_value({name : (__thonny_helper.repr(value), id(value)) for (name, value) in globals().items() if not name.startswith('__')})

STDOUT:

STDERR:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 1, in <dictcomp>
NameError: name '__thonny_helper' isn't defined
```

Here is some simplified code to demonstrate the issue:

```
class Bomb:
    def __init__(self):
        print("running __init__()")
    
    def something(self):
        print("doing something")
        
    self.ooops = True
    
    def otherthing(self):
        print("doing otherthing")
```

self.ooops tries to assign to a property outside a method block. Once, moved into __init__ the error message goes away.

biodives, 2021-05-16:
I just realize that the issue may very well reside with thonny instead of pico micropython. But at least it is now documented somewhere.

aallan, 2021-05-16:
Ping @aivarannamaa

aivarannamaa, 2021-05-17:
This happens because mistakes at class level mess up MicroPython. @biodives  and @aallan, you can help me convincing Damien that this problem is worth fixing: https://github.com/micropython/micropython/issues/7171.

The best I could do in Thonny 3.3.7, was to show a less verbose error message (https://github.com/thonny/thonny/issues/1788). I haven't pinged Simon Long about 3.3.7 yet. (I hope to release 3.3.8 in couple of days, so I think I'll save him from publishing 3.3.7.)

biodives, 2021-05-17:
Thanks for looking into this. My main point of confusion was that the verbose thonny system error message distracted me from the actual triggering error message at the very start. Hopefully the root cause in micropython can be fixed but until that time a more concise thonny error message should help a lot.

closed 2021-05-28 by aallan