raspberrypi/pico-micropython-examples #53: blink.py error

Rudra076, 2022-12-12:
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
TypeError: can't convert str to int

I am using thonny to code and this error is showing up in the shell when I click run current script

lurch, 2022-12-12:
Questions about MicroPython are best asked at https://github.com/orgs/micropython/discussions

lurch, 2023-02-28:
Looking at this again, line 3 of https://github.com/raspberrypi/pico-micropython-examples/blob/master/blink/blink.py says
```python3
led = Pin("LED", Pin.OUT)
```
so I suspect you may simply be using an older version of MicroPython. If you use the latest version from https://www.raspberrypi.com/documentation/microcontrollers/micropython.html#drag-and-drop-micropython that should fix your problem.