raspberrypi/pico-micropython-examples #70: webserver.py NameError: name 'stateis' isn't defined

hairem, 2023-10-10:
I copied and pasted the code from the webserver.py into my pico w and got this error.
```
Traceback (most recent call last):
  File "<stdin>", line 72, in <module>
NameError: name 'stateis' isn't defined
```
if I comment out this line and line 76 I get a different error
```
Traceback (most recent call last):
  File "<stdin>", line 43, in <module>
OSError: [Errno 98] EADDRINUSE
```
Is there any way to fix this?

aallan, 2023-10-10:
This sounds like an issue better suited to [the forums](https://forums.raspberrypi.com/viewforum.php?f=146) where you'll probably get more help than we can offer here.

closed 2023-10-10 by aallan

Thijxx, 2025-03-13:
After a couple of years this is still an issue for a lot of people by the looks of a quick Google, the documentation here https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-python-sdk.pdf still contains the code that will fail a lot with `OSError: [Errno 98] EADDRINUSE` so maybe there should be a note in the document warning people.

This question was floated in the forums; 
 - the suggested fix of using `SO_REUSEADD` does not fix the issue
 - using the reset button does not fix the issue
 - same in PyCharm and VScode

It seems that the only fix is physically reconnecting the device.

lurch, 2025-03-14:
@peterharperuk Any idea what might be (occasionally?) going wrong here?

lurch, 2025-03-14:
Ahhh, I did a quick bit of searching and it looks like you might be running into https://github.com/micropython/micropython/issues/10812 ?