raspberrypi/pico-micropython-examples #78: USB Midi functionality not stable on device removal

ayjaym, 2024-08-23:
1. Using the midi example code run from mpremote as per the instructions
2. Verify that the example logs transmit events over MIDI
3. Connect the Cockos Reaper DAW (most recent version, 7.20) [this can be downloaded for free]
4. Observe that the Board USB device appears in Reaper's MIDI device list
5. Create a track, assigning the board USB device as the input
6. Verify that MIDI events are correctly received
7. Exit Reaper
8. Unplug the Pico from the USB port
9. Plug the Pico back into the USB port
10. Attempt to connect via mpremote. The COM port previously used is now marked 'in use'
11. Attempt to restart the computer. It will hang during restart and require power to be cycled

The current USB implementation appears to leave the device in an undetermined state. Reaper is a very mature and stable product and is highly unlikely to be the cause here. Presumably some event which ought to occur when a USB device is disconnected does not occur for the Pico, leaving resources locked and causing shutdown and port acquisition to both fail.

lurch, 2024-08-24:
AFAICT this repo doesn't contain any MIDI examples, so I suspect that https://github.com/orgs/micropython/discussions would be a **much** better place for your question.

ayjaym, 2024-08-25:
My bad, sorry. There is another repo with examples which were produced for the new USB functionality in 1.23 and I need to locate that and post there. Sorry about that.

closed 2024-08-25 by lurch