raspberrypi/pico-micropython-examples #79: Raspberry Pi Pico ADC changes the sampling rate in Free-running Sampling mode

leezisheng, 2024-09-09:
I am configuring the ADC sampling rate here. Due to unknown reasons, the DIV register cannot be configured here. The actual sampling period is fixed at 65.
Can anyone help me what is going wrong here？
[image]
[image]
[image]

leezisheng, 2024-09-09:
[image]

leezisheng, 2024-09-09:
Solved, The right hand expression div_value | (int_part << 8) | frac_part seems wrong.

lurch, 2024-09-09:
This is probably something that you should ask about at https://github.com/orgs/micropython/discussions rather than in this repo.

leezisheng, 2024-09-09:
done,expression` div_value | (int_part << 8) | frac_part `change to ` (int_part << 8) | frac_part`.

lurch, 2024-09-09:
For anybody else finding this issue in future: please see https://github.com/orgs/micropython/discussions/15816

closed 2024-09-09 by lurch