raspberrypi/pico-micropython-examples #80: PICO2 with python case SPI SLAVE MODE

bawanqin1987, 2024-10-23:
hi sir
  i wanna code the py case with spi slave mode on PICO2,and the other soc board spi with master mode and send data
but pico2 receive data error,all data is 0xff,
[image]

and,i catch the signal with my soc board,the signal data as below:
[image]

please make sure  the py case support spi slave mode?
[image]
[image]

bawanqin1987, 2024-10-23:
from machine import Pin, SPI
import time

# 初始化SPI对象
spi = SPI(0)

# 配置SPI参数
spi = SPI(0, 1000000, polarity=1, phase=0, bits=8, firstbit=SPI.MSB, sck=Pin(2), mosi=Pin(3), miso=Pin(4))

# 配置片选引脚
cs = Pin(5, Pin.IN)
# 短暂延迟，等待接收数据
time.sleep(2)
# 设置引脚输出高电平

# 接收数据
#received_data = bytearray(512)
#time.sleep(5)
# 拉低片选引脚，选择从设备
cs.value(0)
    
# 发送空字节以触发从设备的数据传输
spi.write(b'\x00')
received_data = spi.read(512)
cs.value(1)
time.sleep(1)
# 打印接收到的数据
print("Received data:", received_data)

lurch, 2024-10-23:
This isn't related to the examples in this repo. You should ask for more help at https://github.com/orgs/micropython/discussions an/or https://forums.raspberrypi.com/viewforum.php?f=146

closed 2024-10-23 by lurch

lurch, 2024-10-23:
P.S. I found a comment https://github.com/orgs/micropython/discussions/10442 from January 2023 saying that Micropython doesn't yet support SPI slave mode.